ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #1: no z axis output

cuca94, 2019-10-27:
Hi again!

I am using your code with an AWR1642 board on SDK3.3.
Running fine so far, but im not getting any output on the z axis. I thought the board isn't capable of calculating the z data, but now i read the obstacle detection example where they use the 1642 for 3D output...

is there a way to get z axis data with the demo program?

Link to the obstacle detection guide:
[](http://dev.ti.com/tirex/content/mmwave_automotive_toolbox_2_7_1/labs/lab0004_obstacle_detection/docs/ODSdemo_usersguide.html)

ibaiGorordo, 2019-10-27:
Hi,

I guess you were using the configuration I uploaded to this repository. I forgot to set it to 3D, so it was only using the 2 horizontal transmitter antennas, that is why there was no Z data. I have updated the configuration file to use the 3 transmitter antennas so it should work now. 

However, I do not have the board here, so I am not sure if I edited the configuration correctly. If it gives you some error, go to the demo visualizer from TI and generate a proper configuration file.

Hope it fixes your problem,
Ibai

cuca94, 2019-10-27:
I think your edit is only for the AWR1843 because und the AWR1642 has just 2 Tx antennas...

ibaiGorordo, 2019-10-27:
Yeah, sorry. I forgot that you were using the AWR1642BOOST board. With the normal  AWR1642BOOST board it is impposible to get 3D data because the antennas are in the same horizontal level. You need to have some of the antennas at different heights (like in the AWR1642 "ODS" board) to get 3D position.

cuca94, 2019-10-27:
yeah, that's what i thought..thank you

closed 2019-10-27 by cuca94