ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #2: Question about Matlab demo source

DaltonWoo, 2019-11-10:
Hi, I am using the AWR1843 for a project, but I would like to read the raw data into Matlab for signal processing (as the radar integrates with other modules currently using Matlab), and I was wondering where you found the Texas Instruments Matlab demo. From searching their website, it does not seem like they have a Matlab demo available. I would greatly appreciate it if you could provide a link to the demo that you found. Thank you very much for your time.

ibaiGorordo, 2019-11-11:
Hi, 

At teh beginning I started form the zone occupancy demo for the IWR1443. I then edited the code to also work with the SDK 3. I have just uploaded a folder with my Matlab code slightly edited. I have not tested this version, but I think it should work correclty.

Let me know if you have any problem, (remember to change the serial ports on the "radarSetup18XX"   file).

Best regards,
Ibai

DaltonWoo, 2019-11-11:
Thank you for the fast reply and for the code. I really appreciate it :)

DaltonWoo, 2019-11-13:
Hi, I have changed the serial ports and tried running the code, but I am currently getting this error:

`Index` exceeds the number of array elements (65536).

Error in readAndParseData18XX (line 108)
            tlv.type = sum(byteBuffer(idx+(1:4)) .* word);`

I was wondering if you could provide some insight. Thanks again!

ibaiGorordo, 2019-11-18:
Hi, sorry for the late response. I have made some small changes and the program works with my board. 

One thing I corrected was that the **shiftIndex** variable in "readAndParseData18XX.m" had the value of idx. In the case that a message received from the radar is different from the three types of messages that are parsed, then shiftIndex will have a wrong value and then the next data parsing could be wrong.

I am not sure if that was the problem, let me know how it goes. If your receive the same problem print the value of idx to see if it it some strange value.

Ibai

DaltonWoo, 2019-11-19:
Hi, it runs on my board now. Thanks again!

DaltonWoo, 2019-11-20:
Hi, I have been looking at the Range Profile and after modifying readAndParseData18XX to return rp instead of detObj, I have been receiving this error:

```
Output argument "rp" (and maybe others) not assigned during call to "readAndParseData18XX".
```

I have been checking the value of tlv.type and it never returns 2, so it never enters the case statement for the Range Profile. I am not sure why this is the case. It would be awesome if you could provide some help. Thanks :)

ibaiGorordo, 2019-11-20:
Hi, what configuration file are you using? My first guess will be that it is not configured to send the range value but I am not sure. If you want to remove that error initialize the rp variable at the beginning of the function ( rp = []).

However, that will not solve the problem of not received those packages.

DaltonWoo, 2019-11-20:
Oh I didn't realize that the range value would need to be configured.  I'm using the configuration file that you provided for Python. Is it configured for your config file? I'm not sure how to check.

ibaiGorordo, 2019-11-20:
My config file only sends the point cloud data. You can get the range data by changing the line of guimonitor in the config file to "guiMonitor -1 1 1 0 0 0 0".

You can also generate a new configuration file using the demo visualizer from Texas Instruments.

Best regards,
Ibai

DaltonWoo, 2019-11-20:
It works! You have been a really big help, thank you so much :)

closed 2019-11-20 by DaltonWoo