ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #5: advanced frame configuration

FlorianMaier123, 2020-09-17:
Hey,

i got my sensor today and tested your code. Its working for the demo configuration, but as expected, I get an error when I use the advanced frame configuration. (I used the example cfg from the SDK) The error is:
   startFreq = int(float(splitWords[2]))
ValueError: could not convert string to float: 
Do you have an idea how to solve it?

Thank you
Florian

ibaiGorordo, 2020-09-17:
Hi, 

Right now I do not have time to look into it, but in this SDK there is no real need to read the configuration file, so you can probably comment the line 274 and I think it should work. 

You will have to modify the code so that the data is stored in a different matrix depending on the subframe, or you can store also the subframe in the output dictionary.

I hope that makes it work, let me know if you cannot do it.

Ibai

FlorianMaier123, 2020-09-17:
I commented the "configParametes" and its running now. Is there a reason this part is still in the code? Or is it only for the SDK 2? 

Thanks a lot!

ibaiGorordo, 2020-09-17:
I am glad to hear that. It is mostly because this repository was built on top the SDK 2 and in some cases part of the config data is needed (for example the other script for the Range Doppler heatmap).

Ibai

closed 2020-09-29 by ibaiGorordo