ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #6: Azimuth angles config and code

aghyad97, 2020-09-29:
Hello Ibai, 
Thank you for the great work and this code Ibai.

I am wondering if you will be releasing the code on azimuth angle anytime soon.

If not, can you please give some hints on how to implement it or any good resources?

Thank you.

ibaiGorordo, 2020-09-29:
Hello,

do you mean the range-azimuth heatmap? or do you mean the azimuth angle of the points?

In the case of the first one, I am not planning on implementing it (it would take too much time), but I can give you some info to do it on your own.

In the second case, you can directly extract it like this:

`azimuth = np.arctan2(detObj["y"], detObj["x"]) * 180 / np.pi`

I have not tested but I think it should work, if you want to flip the angle direction then modify the code to this:
`azimuth = np.arctan2(detObj["y"], -detObj["x"]) * 180 / np.pi`

Ibai

aghyad97, 2020-09-30:
First of all, I really appreciate the quick reply, WOW!
Second, I'm actually interested in the first case which is the range-azimuth heatmap, so I would really be thankful if you can provide me with some info so I do it on my own.
For the second case, I will test it out and share my experience.
Thank you a lot Dr. Ibai.

ibaiGorordo, 2020-09-30:
Hi, 

I have not tested this repository, but they seem to be able to provide Range Azimuth heatmap for the IWR1843, so I think it is your best option:
https://github.com/m6c7l/pymmw 

You can either use the repository itself, or if you want a simpler program, you can take the plot Range-Azimuth heatmap part and combine it with this repository.

If it does not work, here (https://github.com/ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2/blob/heatMapDevelop/heatMapDemo.py) is my code trial to get the Range-Azimuth, but the results are not correct (I took the code from the source code in the demo visualizer but I probably missed something). You can probably check the previous repository and correct the code.

And finally, you can also check the source code in Javascript (with some Matlab comments) of the demo visualizer.

Good luck!

aghyad97, 2020-10-02:
I checked the toolbox you mentioned, it seems interesting and has what im looking for.
If it doesn't work, I will go with your code and fix it hopefully.
Thank you a lot for the help, really appreciated.
Aghyad.

aghyad97, 2020-10-10:
Hi again,

So I was looking on detected objects and there features x, y, z and velocity. So, I am getting values for x, y and velocity however, z is always zero. I tried to test it with different configurations but still 0. 

Also, i'm checking tlv_type values and I am getting the right results in regards to my configs.

Any ideas of what might be causing the problem?
Thank you.

ibaiGorordo, 2020-10-10:
Hi, 

Just to make sure, what board are you using?

Do you get the same results with the demo visualizer?

Ibai

aghyad97, 2020-10-10:
I am using AWR1843, and I tried the same configuration on demo visualizer and it's working fine.

Also, I have used both of the config files you provided in your code and got same results, 0 for z.

Aghyad.

ibaiGorordo, 2020-10-11:
That is really weird, the program sends directly the configuration parameters without doing as in the file, and then I since you are getting the velocity correctly, it is not a problem of the data parsing.

Can you write the following code in line 211, (before the start of the loop) to see what data is in the buffer?
print(byteBuffer[idX:idX + 16*numDetectedObj])

aghyad97, 2020-10-11:
> Here is screenshot after entering this line `print(byteBuffer[idX:idX + 16*numDetectedObj])`
> 
> [image]
> 
> Also here is screenshot for detected object 
> 
> [image]

> As you can see Z is 0 and velocity in this time is zero because there is nothing moving in front of the radar, however, when there is something moving the last values of the array change which are the moving objects like below
> 
> [image]

ibaiGorordo, 2020-10-12:
Hi,

The data parsing seems to be fine, but I am not sure why the z value is zero. I have just tested with a IWR6843AOP and in my case the z value is not 0. I guess it is a problem with the configuration file but I have no idea why you get different results with the demo visualizer.

What SDK version are you using?

aghyad97, 2020-10-12:
I am using SDK 3.4

Also, using this version allows me to see 3d plot in the demo visualizer so Z must be there.

ibaiGorordo, 2020-10-12:
I was asking because I have not tested with the newer SDK versions (currently I can only use the SDK 3.02 for the AOP). Maybe you can try to install an older SDK version for example 3.03 to see if the same error happens. 

Otherwise, I have no idea what is going on.

aghyad97, 2020-10-13:
I have tested on many different SDKs and still getting zero.

Could you please send me your config if it's different than what you have in the repo, because I have found on TI forums that some people facing same issue. 

I have tried to change number of TX and also chirpCfg but still zero.

ibaiGorordo, 2020-10-15:
Hi, sorry for the late response.

I have looked at the configuration file, and there might be a problem with my configuration. I have realized that the chirpCfg only activate Tx1 and Tx3. You can try to replace the chirp lines with the following:
```
chirpCfg 0 0 0 0 0 0 0 1
chirpCfg 1 1 0 0 0 0 0 4
chirpCfg 2 2 0 0 0 0 0 2
```

However, probably the best option is to generate the configuration file from the demo visualizer and use that configuration with this repository.

I hope that will fix your problem.
Ibai

aghyad97, 2020-10-20:
Sorry for the late reply,

I have tested it with the new config you provided and now it's working.

Thank you so much.

ibaiGorordo, 2020-10-22:
Hi,

I am glad to hear it. I will fix the config file. 

Thank you, and let me know if you need anything else (I will close this issue).

closed 2020-10-22 by ibaiGorordo

Ffalzter, 2022-03-21:
Hey guys,

Good work!
I am looking for a script to parse the Azimuth/Elevation static heatmap of my IWR6843AOPEVM using mmWave SDK 3.5. .
@ibaiGorordo does you parser you poste above also work for current SDK versions?

Best regards

ibaiGorordo, 2022-03-21:
Hi,

you mean the azimuth-range heatmap, right? As I mentioned above, I have not worked with the azimuth heatmap, so I do not have code for it. If you check the library I mentioned above (https://github.com/m6c7l/pymmw/blob/master/source/app/plot_range_azimuth_heat_map.py), you might be able to extract the logic to draw the heatmap.

Ffalzter, 2022-03-21:
Hey,

thank you very much for your fast response!

Yeah I mean the azimuth-range heatmap (TLV Type 8 Elevation/Azimuth Static Heatmap - found here [OOB Uart Data](https://dev.ti.com/tirex/explore/content/mmwave_industrial_toolbox_4_10_0/labs/Out_Of_Box_Demo/docs/understanding_oob_uart_data.html))

You have published a Heatmap Demo (see https://github.com/ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2/blob/heatMapDevelop/heatMapDemo.py). This can be used to parser the necessary data to plot it in the next step, right?

Best regards

ibaiGorordo, 2022-03-21:
My code was for the range azimuth heatmap (you can check the value of the TLV type was 4 as in the range azimuth). And I have never been able to get good results. That processing was obtained from the official demo visualizer app with some conversion, but I probably missed something.

That Azimuth Elevation seems to be new, I had not seen it before and it is not available in the demo visualizer and I cannot find any example using it. I am also not sure how the configuration would look like.

So, I am not sure where you could start looking. I actually have one AWR1843AOP available, even though it is not compatible with the Industrial toolbox, since it is for the Out of the box demo I might be able to test it. Let me know if you find any example in Matlab or Demo Visualizer using it and I will try to check it (it might take time).

ibaiGorordo, 2022-03-21:
I will open an issue so that we can continue talking there.

Ffalzter, 2022-03-21:
In order not to confuse, the URL I posted before shows the UART output format of the Out of the Box demo, so it is not just for/from the industrial toolbox. But yeah, to have azimuth AND elevation you need a sensor with the required number of antennas.
As you pointed out above, your code is for the range azimuth heatmap what should be the same (doppler = 0 -> just static objects).
See you in the new issue.