ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #7: byteVec sometimes empty

FlorianMaier123, 2020-10-05:
Hi,

I read AWR1843 with diffrent configurations but I have always the same problem. Sometimes (about every 10th time) the byteVec just stays empty. I testes it on Mac and Windows. On Windows, sometimes the program starts reading and after a few seconds the byteVec is empty again. I changed the frame rate to 2 frames/s but the problem is the same. You have an idea, what else I could try?
Thanks for your help!

ibaiGorordo, 2020-10-05:
Hi, 

I am not sure if that is the problem but the current program expects around 20 FPS, with a much slower rate, you should probably increase the sleep time to 0.5 seconds in the case of 2 FPS.

Ibai

FlorianMaier123, 2020-10-05:
Hi,

I tested different values for the frame rate and the sleep time, but the problem is the same. You have another idea?

FlorianMaier123, 2020-10-05:
I'm using the pyserial package instead of the serial package. Could this be a problem?

ibaiGorordo, 2020-10-06:
In the code the Pyserial is called serial, so you are using the correct package. I am not sure what might be the problem.
Some things to consider:

-What configuration are you using?
-When you say the byteVec is empty, what does it happen? I mean does it crash?
-How long does it stay empty? You say that 1 out of 10 times is empty, but even if it is empty there should be no problem because the code should skip until new data is available.
-Do you see a similar behavior when you use the same configuration in the demo visualizer?

FlorianMaier123, 2020-11-24:
Hi,

sry for my late response. I decided that the TI sensors are not the right sensors for my application. But I still want to describe what the problem was.
I tried a lot of different configurations (for example yours or the one of the demo visualizer).
The program did never crash but sometimes the byteVec was just an empty variable. If this occured, it remained empty and didn't get new data after some time. When I use the demo visualizer it's always working.

But like I said, it's not an important topic anymore. Thanks for your previous help.
Florian

kongto-yu, 2024-05-26:
Hi,

Thanks for your code first. I deployed the mmWave radar system based on your code. Could you please explain why there should be a sleep operation after the main parts?

Best,
Luca