ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #9: z-axis values are all zero

anathema43, 2020-10-27:
When I print the objdetect all the z values are 0.

Also when I tried to give a configuration.cfg saved from MMDemo visualizer the plot screen is blank

Any inputs .
Thanks

ibaiGorordo, 2020-10-27:
Hi,

Please, let me know about more information about your setup:

- Board
- SDK installed
- Does the 3D scatter work in the Demo Visualizer?

Thank you,
Ibai

anathema43, 2020-10-28:
@ibaiGorordo thanks for the quick reply.

board- AWR1843BOOST

SDK installed - I never flashed any new sdk, but from demo visualizer, i understand its 3.2

demo visualizer works please find the screenshot attached
[image]

the output of  readData_AWR1843.py
[image]

As you can see the z-axis is always zero. 
and if I try to save a config from demo visualizer and load it,  gives a bank screem\n

ibaiGorordo, 2020-10-28:
Hi, 

Thank you for the detailed information, the zero value might be because my config file is not correct. The problem is that I do not have the AWR1843 anymore so I cannot test if the config file works.

But I do not understand why the demo visualizer config does not work. When you run the config file from the demo visualizer, does the radar start measuring? (it should do some noise and some LED should turn on). 

If it does not, I think there is a problem with the configuration file because the program only reads the txt file and sends the configuration as it is in the file. Also, since you took it from the demo visualizer, the configuration file should be fine, so, I am not sure what might be the problem. 

Also, in the meantime, you can try to flash a newer SDK in the board and see how it goes. But, honestly I do not think it is going to change,

Let me know how it goes.
Ibai

anathema43, 2020-10-28:
hey @ibaiGorordo 

thanks for your reply. 
I now can get the values in z-axis while uploading the configuration file from mmDemo Visualizer. I am not sure y it happened now but I changed the antenna config of  "Antenna config(Azimuth Res-deg) to 4Rx3Tx(15 deg +Elevation) " in the demo visualizer.
before it was  4Tx2Tx and as the script has hard-wired Tx and Rx I suppose that was the conflict, I suspect, but i am new to mmwave.

Also, i would like to ask the (x,y,z) of objdetect  is the z-axis depth, y is the height?

what the difference between 3d scatter plot and range profiler in demo visualizer?

thanks

ibaiGorordo, 2020-10-28:
I am glad to hear that.

Y is depth and Z is up, so the plot in this repository shows the top view of the positions of the points.

The 3D scatter plot is the 3D position of the detected points after the CFAR algorithm, whereas the range profiler profiler shows the power at different range intervals for the zero doppler (the peaks detected should correspond with the range of the detected points).

https://www.ti.com/lit/ug/swru529c/swru529c.pdf?ts=1603882653464&ref_url=https%253A%252F%252Fwww.google.com%252F

anathema43, 2020-10-28:
Thanks for your work and help. I now close this issue.

closed 2020-10-28 by anathema43