ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #10: The value of detObj.z is always zero

Carbord, 2020-11-03:
Hello~ I encountered the problem that the value of detObj.z is always zero while running this code of AWR1843. I wonder if there exists any way to fix this problem.

ibaiGorordo, 2020-11-03:
Hi, 
Probably is a problem with my configuration. Have you tried using a configuration file downloaded from the Demo Visualizer Studio?

Carbord, 2020-11-03:
> Hi,
> Probably is a problem with my configuration. Have you tried using a configuration file downloaded from the Demo Visualizer Studio?

Thanks, I will try soon . Did you ever got non-zero z-coordinate in your experiments ? By the way, I wan to know that which configuration parameters will decide the point number,  since I got only 10 points in a frame while using the configuration in this repository.

ibaiGorordo, 2020-11-03:
Lately I have not been able as I no longer have the AWR1843 board, but I am using the same code in the IWR6843AOP and it work well (with a different configuration file).

As for increasing the number of points, you can remove the peak grouping options in the Demo visualizer and reducing the CFAR thresholds. One note about this, when you save the configuration file in the Demo Visualizer, if you change the peak grouing option or the CFAR threshold, it will not be updated in the .cfg you download. You will have to manually modify the lines in the .cfg file. You can use the commands that are displayed on the Demo Visualizer to see what commands to copy.

closed 2021-12-15 by ibaiGorordo