ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #11: Implementing other package types

AydenBurgoyne, 2020-12-04:
Hey I'm wondering if you could provide some resources that could help me implement the other package types such as range profile, etc. Thanks

ibaiGorordo, 2020-12-06:
Hi, 

In that case you will have to add the code after the line 227. If you want the range profile info you can add something like this:
```
 elif MMWDEMO_UART_MSG_RANGE_PROFILE:
     # Get the number of bytes to read
     numBytes = 4*configParameters["numRangeBins"]
     rangeProfile = byteBuffer[idX:idX + numBytes ].view(dtype=np.float32)
 
```

I have not tested but I think it is correct.

Let me know if you have any question.
Ibai

anathema43, 2021-01-08:
hi, i wanted the range profile, i tried the above code and it never enters the elif condition. could you please provide some more guidance on how to implement it?

ibaiGorordo, 2021-01-08:
You can try to ```print(tvl_type)``` on line 202 to see what types of data you are receiving.

Also, did you modify the config file? You should modify it because the current config only gets the 3d point cloud data.

Ibai

closed 2021-12-15 by ibaiGorordo