ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #13: Collecting data from Sensor AWR1642

RanjaniN-10, 2021-03-08:
Hi,
I'm using your python code for collecting data like range, velocity from AWR1642 Radar. I have replaced the config file of the python code with config from the mmwave demo visualizer. 

I could visualize the range in DemoVisualizer (refer image) 

[image]

I am  able to get the velocity using your code, but not the range(in meters).

[image]
Could you please help me with finding the range?

System Configuration:
OS type: Ubuntu 18.04, 64bit
Processor: Intel Core i7-7700 CPU @ 3.60GHz × 8
RAM: 16GB
GPU: GeForce GTX 1080 Ti

Thanks in advance
Ranjani N

ibaiGorordo, 2021-03-08:
Hi, are you interested on the range profile like in the demo visualizer, or are you looking to get the range of the detected points?

RanjaniN-10, 2021-03-08:
Hi, 
Thanks for your reply. We need to know the range of the detected point in meters.

ibaiGorordo, 2021-03-08:
In thay case it should be easy, you just need to do sqrt(x^2+y^2+z^2).

I am writing with the phone so I can't give you the exact code, but you should be able to easily calculate it using numpy.

RanjaniN-10, 2021-03-08:
Hi @ibaiGorordo 

Thanks for the reply, found the equation for calculating range and successfully implemented using numpy library.

ibaiGorordo, 2021-03-08:
Glad to hear that.

Just for future reference I am writing the code here:

`range = np.sqrt(detObj["x"] ** 2 + detObj["y"] ** 2 + detObj["z"] ** 2) `

(it could be possible to use the numpy norm function by combining the arrays into a 2D array)

closing thread.

closed 2021-03-08 by ibaiGorordo