ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #16: mmWave-SDK-3.5 installation in raspberry pi-4

Malathi15, 2021-03-12:
Hi @ibaiGorordo 
Thanks for your support. I have installed mmWave-SDK-3.5 and successfully run the AWR1642 Boot python code in my two Ubuntu Systems. 
System configuration:
```
OS type: Ubuntu 18.04, 64 bit
Processor: Intel Core i7-7700 CPU @3.60GHz x 8
RAM : 16GB
GPU: GeForce GTX 1080 Ti 
```
```
OS type: Ubuntu 18.04.5 LTS, 64 bit
Processor: AMD Ryzen 7 3800x 8 Core Processor x 16
GNOME : 3.28.2
GPU: GeForce GTX 1080 Ti
```
I installed `ubuntu-18.04.5-preinstalled-server-arm64+raspi3.img.xz` on my Raspberry Pi4, which is 64 -bit Ubuntu 18.04.
While I have tried to install mmWave-SDK-3.5 I got an Error ` cannot execute binary file: Exec format error` (Please refer the image).
[image]

I was not able to install the packages discussed in this [Link](https://e2e.ti.com/support/sensors/f/sensors-forum/759934/linux-mmwave-sdk-cannot-install-mmwave-sdk-in-linux). While trying, I faced the Error  `E: Unable to locate package`.
So as a work around I tried installing the packages: libc6-arm64-i386-cross from synaptic package manager.Still the error persists.

Could you please help with install mmWave-SDK-3.5 in raspberry pi-4

Thanks in Advance
Malathi K

ibaiGorordo, 2021-03-12:
Hi, 

You do not really need to install the mmWave SDK into the Raspberry Pi. That is more for compiling your own firmware, or programming the firmware into the radar. You can do that from your computer.

In the Raspberry Pi, you just need to install Python and the required Python libraries because the radar is already preprogrammed to send and receive data through the Serial port without the need to install any driver.

Malathi15, 2021-03-12:
@ibaiGorordo Thank you for your reply

closed 2021-03-12 by Malathi15