ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #18: Estimation of Angle of Arrival(AoA)

Malathi15, 2021-03-18:
Hi @ibaiGorordo 
I am using AWR1642. I would like to calculate **Angle of Arrival(AoA)** using your python code` readData_AWR1843.py`
The objective of trying to calculate AoA is to know if the obstacle is  arriving to the left or right of the sensor.

The logic I had is that the X-coordinate is negative in case of obstacle being arrived from the left and  positive in case of obstacle being arrived from the right. But here the angle is not estimated here. It would be more precised to use the angle.

Q1: Is my thought process of finding the position(left/right) correct?
Q2: How to estimate the AoA?

Thanks in advance
Malathi K

ibaiGorordo, 2021-03-18:
Hi, you just need to do `np.arctan(detObj["x"]/detObj["y"])*180/pi`

Make sure that the sign of X matches your logic, so that when it is negative, the angel is going to be negative

Malathi15, 2021-03-24:
Thanks for your Reply @ibaiGorordo .

closed 2021-03-24 by Malathi15