ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #23: Issues with pyqtgraph UI

HaidStefan, 2021-06-28:
Hi, 

can you please help me with the specific versions you are using?
I tried out a lot of different constellations regarding the pyqtgraph package and dependencies but couldn't get the UI running. 

I tried Python 3.6 / 3.7 with Psyside2, Pyside6 or PyQt5.
I either have problems with deprecated components, freezing UI or DLL load failures.

Can anyone please give me the specific versions of each dependency and the Python Version that is working for you?

`
python --version
pip freeze
`

Would suggest to add a requirements.txt `pip freeze > requirements.txt` to make it easier for others installing the required packages.

Thanks a lot in advance
Stefan

ibaiGorordo, 2021-06-28:
Hi,

I will not be able to test the radar for a couple of days, so I do not have the specific versions, but I have never have a problem with one of the latest versions of the library. Normally I install pyqt5 and then, pyqtgraph. In the meantime, I would suggest to try to create a new python environmnet and install the packages one more time.

What error do you get? Are you doing it in a computer? or in some SBC?

HaidStefan, 2021-06-29:
Hi,

i tried again today and i got it working now.  I had the following warning yesterday:
`C:/workdir/github_mmwave/AWR1843-Read-Data-Python-MMWAVE-SDK-3-/readData_AWR1843.py:281: DeprecationWarning: GraphicsWindow is deprecated, use GraphicsLayoutWidget instead,will be removed in 0.13
  win = pg.GraphicsWindow(title="2D scatter plot")`

and i thought that this is the reason the window was freezing.
Actually the reason was that I had missmatches in the SDK Version of the configuration file and the board.

Thanks a lot for your help!

closed 2021-06-29 by HaidStefan