ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #24: Determination of Z - Coordinate

Malathi15, 2021-07-09:
Hi,
I'm using https://github.com/ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- to read the data from` AWR1642 EVM`.
I tried to find the height of the detected object by excuting readData_`AWR1843.py`, using `Z` coordinates. The `Z` values are always zero.
Here I have attached the config file created from Demo Visualizer. 
```
sensorStop
flushCfg
dfeDataOutputMode 1
channelCfg 15 3 0
adcCfg 2 1
adcbufCfg -1 0 1 1 0
profileCfg 0 77 438 7 48.98 0 0 15 1 256 6250 0 0 30
chirpCfg 0 0 0 0 0 0 0 1
chirpCfg 1 1 0 0 0 0 0 2
frameCfg 0 1 16 0 100 1 0
lowPower 0 1
guiMonitor -1 1 1 0 0 0 1
cfarCfg -1 0 2 8 4 3 0 15 1
cfarCfg -1 1 0 4 2 3 1 15 1
multiObjBeamForming -1 1 0.5
clutterRemoval -1 0
calibDcRangeSig -1 0 -5 8 256
extendedMaxVelocity -1 0
bpmCfg -1 0 0 1
lvdsStreamCfg -1 0 0 0
compRangeBiasAndRxChanPhase 0.0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0
measureRangeBiasAndRxChanPhase 0 1.5 0.2
CQRxSatMonitor 0 3 4 127 0
CQSigImgMonitor 0 127 4
analogMonitor 0 0
aoaFovCfg -1 -90 90 -90 90
cfarFovCfg -1 0 0 50.00
cfarFovCfg -1 1 -1 1.00
calibData 0 0 0
sensorStart
```
1) Can you please let me know how to determine the Z coordinate?
2) Is it possible to determine the height of objects using Z coordinate?

Thanks in Advance
Malathi K

ibaiGorordo, 2021-07-09:
Hi Malathi,

The AWR1642EVM has all the antennas in the same height, so, it cannot differentiate object position in the z axis. That is why it returns z=0. You will have to use some of the other XWRXX43 chips or the AWR1642ODS to estimate height of objects.

Regards,
Ibai

Malathi15, 2021-07-10:
Hi @ibaiGorordo Thanks for your reply
Can you please clarify the following doubts
1)  Can I use the same code to test` AWR1642BOOST-ODS EVM`?
2) Can you please let me know, if `AWR1642BOOST EVM & AWR1642BOOST-ODS` have the same range?

ibaiGorordo, 2021-07-10:
Hi,

> Can I use the same code to test AWR1642BOOST-ODS EVM?
Yes, the code only changes with the SDK version, but not the hardware. You will probably have to modify the config file to suit your board though.

> Can you please let me know, if AWR1642BOOST EVM & AWR1642BOOST-ODS have the same range?
They should probably have the same range since you are only changing the antenna configuration and not the chip itself, the angular resolution will be lower though. However, it is better if you ask these question in the TI forum.

> Are there any advantages in increasing the number of frames in DemoVisualzier config? 
Do you refer to the frame duration? It is going to depend on your desired application. However, I recommend you to use the demo visualizer to test different values, and once you got a configuration you like, you can save it and use it with this code.

Malathi15, 2021-07-12:
@ibaiGorordo Thanks for your response. 

Thanks and Regards 
Malathi K

closed 2021-07-12 by Malathi15