ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #26: USING AWR1642BOOST-ODS

guruvishnuvardan, 2021-10-05:
Hi Ibai,

Good Day!

Can you please let me know, if it's possible to use this code to analyse the performance of AWE1642BOOST-ODS EVM?

As of now, we are using AWR1642BOOST EVM, Since we need elevation data(+ & - of elevation data) to determine the obstacle from a specific height (Meaning - from 1 feet above the ground to 10 feet)

Thanks
Guru

ibaiGorordo, 2021-10-05:
Hi,

The code will work with the ODS version as long as it has the SDK version 3 installed (otherwise use the repository for the installed version).

If you need the elevation probably the ODS model, the 1843, or even the AWR1843AOP versions will work. As for the height limits, you will probably have to check based on the field of view of each device.

guruvishnuvardan, 2021-10-06:
Hi Ibai,

Thanks for your response.

Please find the FOV details:

AWR1843 -- Integrated three-transmit four-receive (3TX/4RX) AoP with 140° Azimuth field of view (FoV) and 140° elevation FoV

AWR1642ODS -- Azimuth: ±70 degrees , Elevation: ±40 degrees

Can I use any of these radars for testing elevation (Example -- 10 Feet above the ground to 0.5 feet above the ground)

Thanks
Guru

guruvishnuvardan, 2021-10-07:
Thanks Ibai. One Curious Question -- During our analysis with AWR1642BOOST EVM, I see, If a pedestrian is detected at 0 Degrees and Car at 0 Degrees is detected at say 5 Degrees sometimes and at 0 Degrees sometimes, is the detection angles depending on the width of an Object?

Thanks
Guru

ibaiGorordo, 2021-10-07:
I guess it will depend also on the distance to the objects. If the car is close, then, it is very likely that points from the car will be reflected at different angles. 

If you can "simulate" the scene setup, you might want to check the azimuth-range heatmap to get a better understanding. You will probably see that the area of the car is bigger.

closed 2021-10-07 by guruvishnuvardan