ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #35: Does this code work with IWR6843AOP

HichriAyoub, 2022-08-05:
Hello,
 i am trying to use this code with IWR6843AOP board with sdk 3.05 the code works fine but gets stuck after sensor start it just reads the config file and sends it and does nothing

ibaiGorordo, 2022-08-05:
Yes it works, but you need to modify the config file to one that is suitable for the AOP board.

You can get it from the demo visualizer. If the config you create works in the demo visualizer, it should also work here.

HichriAyoub, 2022-08-05:
somehow the config file works with demo visualiser sdk 3.5 but when try to run range-dopplerheatmap.py the code stop at sensorstart and i printed dataOk variable it's always 0 i think the problem might be coming from the readAndParseData18xx function but i don't know where exactly do you have any idea ?