ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #36: AWR1843 VELOCITY CALCULATION

guruvishnuvardan, 2022-08-24:
Hi Ibai,

Good Day!

It's been a while, I hope you are doing fantastic!

Can you please let me, if there is any way we can calculate the real velocity from AWR1843BOOST Radar? , I just wanted to differentiate static and moving obstacles and false positives(noises that confuses me, by looking like real data).

Thanks
Guru

ibaiGorordo, 2022-08-24:
You can get the velocity using detObj["velocity"]. However, if you want to remove static noise, it might be better in the config file to enable clutter removal in the demo visualizer (https://dev.ti.com/gallery/view/mmwave/mmWave_Demo_Visualizer/ver/3.6.0/)
[image]

guruvishnuvardan, 2022-08-25:
Hi @ibaiGorordo ,

Thanks for your response.

Please find the screenshot of the velocity printed using detObj["velocity"] variable, the data received for velocity of the obstacle in the front is -0.491, is it meter / second? ,  But if I convert it's 1.44 Kilometres / Hour

Can you please let me know, if it's the right conversation method?

[image]

Thanks
Guru

ibaiGorordo, 2022-08-26:
I don't remember properly, so I am not 100% sure, but I think it is in m/s.

However, it is radial velocity and in your case, since the radar in the car is moving is going to be relative velocity (the difference in velocity between the truck and the car with the radar). Similarly, since your car is moving, even static objects are going to have non 0 velocity.

ibaiGorordo, 2022-08-26:
The way you did is fine and to get the absolute velocity you will have to add the car's velocity

guruvishnuvardan, 2022-08-26:
Thanks @ibaiGorordo

closed 2022-08-26 by guruvishnuvardan