ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #45: Radar output values

Lilii13, 2023-06-13:
Hello,
First of all, I tested your code with the AWR6843AOPEVM board, and it works like a charm.
I check my static points on the mmWave Visualizer, then I test your code and print the {x,y,z,velocity} data and see that it fits for the x and y.
When I try to move my hand above the antennas, the velocity changes from 0 to different values.
However one problem remains with the z parameter which is always equal to 0, although I move my hand in a way to vary in all three axis, though the z remains equal to 0.
Any clues on how to troubleshoot that please ?
Thanks,
Lilia.

SDUtdXW, 2024-06-15:
Hello,
I encountered the same issue while working with the AWR1843 board. The x and y parameters seem to work fine, but the z-parameter remains constant at 0, even though I move my hand in a way that should vary along all three axes.
Could you share any insights or solutions you found to address this problem with the z-parameter? Your experience could be really helpful.
Thanks,
X17.

SDUtdXW, 2024-06-30:
> Hello, First of all, I tested your code with the AWR6843AOPEVM board, and it works like a charm. I check my static points on the mmWave Visualizer, then I test your code and print the [...]

Hello,
I encountered the same issue while working with the AWR1843 board. The x and y parameters seem to work fine, but the z-parameter remains constant at 0, even though I move my hand in a way that should vary along all three axes.
Could you share any insights or solutions you found to address this problem with the z-parameter? Your experience could be really helpful.
Thanks!

fengdehuaiyuan, 2024-12-25:
you can change cfg file