ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- #48: No ouput

SiriDubbaka, 2024-02-23:
I'm using AWR1843 Device, I have used readData_AWR1843.py. I have changed the port numbers correctly and used the same configuration file provided by you. The device is getting connected but I'm not getting any output, the scatter plot is empty. Can you suggest what to do? Is MATLAB required in anyway, cause I'm directly working on Python.

Liwenlong524998, 2024-04-07:
I had the same problem as you, have you solved it yet?

jvithub, 2024-05-02:
I've managed to run the readData_AWR1843.py and get data from the scatter plot using 1843 AOP sdk version 3.6 w/o the need of MATLAB.
Make sure you download and use the correct configuration file for your specific device from https://dev.ti.com/gallery/view/mmwave/mmWave_Demo_Visualizer/ver/3.6.0/
You can try adjusting the parameters in the visualiser as well to see if that helps output something in the scatter plot.

evileleven, 2024-11-08:
I have the same issue.

I use SDK version 3.5, and I get the 'sensorStart' output but no figure output.

haolin23, 2024-12-04:
I have encountered a similar problem. After I modified the configuration file, I couldn't successfully read the radar data from the serial port during debugging. Is my config file wrong or does the radar need to have its program burned in advance? Thank you for getting back to me.